simonsobs/socs #4: Print feedback after attempting agent initialization

JakeSpisak, 2019-06-04:
When running clients in a jupyter notebook, it would be nice if there was feedback after attempting agent initialization (ie print if the initialization succeeded or failed). Currently no output is printed when running the LS240 agent using MatchedClient in a jupyter notebook

jlashner, 2019-07-24:
By design the matched client `start` function doesn't wait for a task to finish, so it won't provide the result. If you want to wait until the task to finish, you can call `wait` after starting the task which will return whether or not the task was successful, or you can run `status` to check on an ongoing task or process.

kmharrington, 2019-07-25:
For the record, I agree we need something like this. At a minimum having some sort of interactive feedback if it failed to start correctly. We're all using OCS in a mostly interactive mode at this point and the lack of feedback when agents fail to start (or fail in general) is a bit annoying.

You could also imagine _really_ wanting this in the real mode of OCS as well. If we push go on the beginning of house-keeping collection on the receiver at the site and then don't realize for hours/days that it didn't start, that's going to be a massive issue. Also, if the agent fails for some reason during operation, we need to know _when if fails_, not days/weeks later when we're trying to correlate things with our bath temperature and realizing we don't have a record of it.

JakeSpisak, 2019-07-26:
Yeah, I assume it is intentionally designed so that you need to explicitly
call wait after a function, which I am doing now. From a user standpoint,
however, I would note that it would be simplest to call one function that
combines the start and wait functionality. For every function except data
acq/monitor output, I think that all of my use cases have involved calling
wait afterwards. It's a simple thing, but I have spent time searching for
why my script didn't work only to find I missed a 'wait' statement.

Jake

jlashner, 2019-07-26:
Well its designed so that you start a bunch of tasks and have them run asynchronously, and like initialize all of the lakeshores at once without waiting for each one to finish. If you want I can add a function `start_blocking` to the MatchedClient that starts a task and waits for it to finish before returning. 

I think monitoring the status of a process like Katie is suggesting is beyond the scope of the Matched Client, but I agree it's necessary to have some sort of feedback. I think that the interactive web dashboard that Matthew is working on will have that since you can view the status messages live. I also like the idea of having a grafana panel which shows a table of important agent operations and their current statuses, but I'm not sure how feasible that is.

kmharrington, 2019-07-26:
I'm going to re-open this as a standing issue for things we'd like to see in the umbrella of `SO OCS`. 

A `start_blocking` function might work. I'm not super familiar with the code structure here. But if we could make a `MatchedClient` with `(mode = 'interactive') that would block on starts/stops and print error messages if data acquisition failed later, I think that could be a nice fix.