simonsobs/socs #6: Lakeshore 240 Communication Time-out

BrianJKoopman, 2019-06-21:
We recently had two of the Lakeshore 240s time-out during data collection at Yale. Both Agents were found with the following traceback:

```
2019-04-03T20:40:11+0000 acq:1 Crash in thread: [Failure instance: Traceback: <class 'TimeoutError'>: Device timed out
/usr/lib/python3.6/threading.py:916:_bootstrap_inner
/usr/lib/python3.6/threading.py:864:run
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/twisted/_threads/_threadworker.py:46:work
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/twisted/_threads/_team.py:190:doWork
--- <exception caught here> ---
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/twisted/python/threadpool.py:250:inContext
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/twisted/python/threadpool.py:266:<lambda>
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/twisted/python/context.py:122:callWithContext
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/twisted/python/context.py:85:callWithContext
LS240_agent.py:129:start_acq
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/ocs/Lakeshore/Channel.py:164:get_reading
/home/bjk49/git/ocs-prod/venv/lib/python3.6/site-packages/ocs/Lakeshore/Lakeshore240.py:70:msg
]
2019-04-03T20:40:11+0000 acq:1 Status is now "done".
```

The agents were still running, just not acquiring data. Perhaps it would be best to try again a couple of times before giving up all together. I wasn't simply able to just run the 'acq' command again, as it thought it was still running. 

```2019-04-04T16:53:43+0000 start called for init_lakeshore
2019-04-04T16:53:43+0000 init_lakeshore:2 Status is now "starting".
2019-04-04T16:53:43+0000 Initialized Lakeshore: False
2019-04-04T16:53:43+0000 init_lakeshore:2 Conflict: "acq" is already running.
2019-04-04T16:53:43+0000 init_lakeshore:2 Status is now "done".
2019-04-04T16:53:43+0000 start called for acq
2019-04-04T16:53:43+0000 acq:3 Status is now "starting".
2019-04-04T16:53:43+0000 acq:3 Conflict: "acq" is already running.
2019-04-04T16:53:43+0000 acq:3 Status is now "done".
```

I ended up killing the agents and restarting them, which worked to resume data taking without any additional steps.

BrianJKoopman, 2019-06-21:
Steps to recreate this error:
* Start 240 Agent
* Being acquiring data
* Turn off the 240 power supply
    * This causes the device to time-out
* Turn on the 240 power supply
* Attempt to start data acquisition again
    * You should now have the 'acq' conflict error on the Agent, and data acquisition will not begin.

@jlashner, I know this problem is solved with an alternative locking mechanism. Is that documented anywhere? Or can you point to an up to date example?

jlashner, 2019-06-21:
Hmmmm the LS240 is already using the new locking mechanism.... Did you have this problem with the most recent version of the 240 agent? Are you re-initializing the agent after you power it back on?

BrianJKoopman, 2019-06-21:
Ah, nope, I'm on a version from before the TimeoutLock was implemented, will update. Thanks!

BrianJKoopman, 2019-08-16:
Tested with up to date 240 Agent. Problem fixed.

closed 2019-08-16 by BrianJKoopman