simonsobs/socs #32: Cryomech Agent: String conversion crash

BrianJKoopman, 2019-11-26:
We saw this at Penn on a PT420 running off of a CPA1114 compressor.

```
2019-11-26T19:22:05+0000 acq:1 Crash in thread: [Failure instance: Traceback: <class 'ValueError'>: could not convert string to float: 'c9000000'
/usr/lib/python3.6/threading.py:916:_bootstrap_inner
/usr/lib/python3.6/threading.py:864:run
/usr/local/lib/python3.6/dist-packages/twisted/_threads/_threadworker.py:46:work
/usr/local/lib/python3.6/dist-packages/twisted/_threads/_team.py:190:doWork
--- <exception caught here> ---
/usr/local/lib/python3.6/dist-packages/twisted/python/threadpool.py:250:inContext
/usr/local/lib/python3.6/dist-packages/twisted/python/threadpool.py:266:<lambda>
/usr/local/lib/python3.6/dist-packages/twisted/python/context.py:122:callWithContext
/usr/local/lib/python3.6/dist-packages/twisted/python/context.py:85:callWithContext
cryomech_cpa1114_agent.py:199:start_acq
cryomech_cpa1114_agent.py:106:get_data
cryomech_cpa1114_agent.py:166:breakdownReplyData
]
2019-11-26T19:22:05+0000 acq:1 Status is now "done".
```

BrianJKoopman, 2019-11-26:
@JakeSpisak would you mind taking a look at this?

JakeSpisak, 2019-11-26:
@BrianJKoopman Looks like it caught an unwanted letter in one of the outputs. This is a similar issue I had with the CPA1000. Was it working for a while, or did it immediately throw this error?

BrianJKoopman, 2019-11-27:
> @BrianJKoopman Was it working for a while, or did it immediately throw this error?

It worked for ~30 minutes, then this error occurred.

mhasself, 2019-11-27:
It would be good to catch these decoding errors (obviously), and not crash...  but also we should log the full garbled message so we can collect and inspect them and perhaps file a bug report with Cryomech.

On ACT we read the compressors out over an RS232 connection, and once in a while the controller will return garbled responses.  I have observed that this occurs much more frequently in cases where the data would otherwise be out of bounds, e.g. when one of the pressures is below the fiducial range.  So I think there's something wrong with their firmware, such that the output communication channel misbehaves when internal error conditions are active.

JakeSpisak, 2019-11-28:
Good point Matthew. The fact that it worked for a while makes me think we can possibly just ignore the bad data points as they occur, at least for now. So I can write something that checks for non-allowable characters, and if triggered returns no data for that time point and prints the raw compressor output to the logs. Then once we have a few of these bad data points I can email someone at cryomech with the bug. The ACT experience makes me suspicious something similar is going on here: the error I had with our upstairs compressor, which I attributed to the fact that it was a different model, occurred while the input water temp was higher than recommended. @BrianJKoopman do you know if the Penn error occurred when the compressor was operating outside normal conditions?

BrianJKoopman, 2019-12-04:
I believe things were "normal". This occurred just after starting all four compressors on the LATR. Only one failed at the time. They report smooth sailing so far on this cooldown though after restarting the one Agent.

Any progress on patching this?

JakeSpisak, 2019-12-06:
Hmm, okay. Sorry no progress so far - have put this on the back burner, but I'll try and get it done over the weekend.

jseibert575, 2019-12-09:
Created a pull request for a fix to this. I tested it on a compressor here - however, all of our compressors are currently working with the previous image (including the CPA1110 compressors. It's off right now, so I'm thinking the problem in using the agent with it beforehand may have been the high water temp warning). So I tested the error handling functionality by generating fake bad compressor output in the agent file.

JakeSpisak, 2019-12-09:
*Realize I was signed in as Joe this afternoon- just to clarify, I am the single point of contact for this agent.

BrianJKoopman, 2019-12-13:
> Created a pull request for a fix to this. I tested it on a compressor here - however, all of our compressors are currently working with the previous image (including the CPA1110 compressors. It's [...]

Great, thanks for the PR. I just wanted to mention, including any code/process for testing would be great too. That way others can, in the future, run the same test case on updates they produce to verify they still pass.

BrianJKoopman, 2020-01-31:
Resolved by #33.

closed 2020-01-31 by BrianJKoopman