simonsobs/socs #44: Keithly documentation + output I and V

JakeSpisak, 2020-02-19:
1) The Keithly agent documentation in socs is empty and 
2) It would be great to have ocs functions to output in python the current and voltage for a given channel. This should be a quick fix since the driver functions are already implemented.

mhasself, 2020-02-19:
For returning the current and voltage -- the way OCS is supposed to handle stuff like this is that a particular Process or Task would record useful information in the session.data member.  The session is returned by all calls (start, stop, status, wait...).  For the keithley Agent, the relevant operation is "monitor_output", and the user would run something like (in a MatchedClient):
```
    status, msg, session = ps_client.monitor_output.status()
    print('The voltage on channel 1 is', session.data['channel1']['V'])
```

This just requires someone to modify the Keithley agent code to populate session.data.  session.data is underused at present, but this is what it's for.

JakeSpisak, 2020-03-24:
I have implemented this and will submit a PR soon so this issue can be closed

closed 2020-07-29 by BrianJKoopman