simonsobs/socs #45: Automatically log sample heater and still heater outputs

kmharrington, 2020-02-19:
It would be convenient to have the powers going to the two (or even three) LS372 heaters being automatically logged / in the data files somewhere.

kmharrington, 2020-02-20:
Update. There _are_ being logged. However they aren't visible in the InfluxDB version of graphana. The channel names are  `observatory.LSA22HG.feeds.temperatures.sample_heater_out` and `observatory.LSA22HG.feeds.temperatures.still_heater_out`

This might not be an socs request anymore. @BrianJKoopman  how do we get to look at this in Graphana?

kmharrington, 2020-02-20:
Another Update. The sample heater and still heater _are_ available in graphana. They just aren't a timestream. It looks like their values are only recorded when they're changed, so you see them if you make a panel but not on a timestream graph

kmharrington, 2020-02-21:
Update again!

The heater values are only recorded when they're changed. We should probably update them more continuously, since eventually we'll be servoing the focal plane times.

Also, in it's current configuration, graphana is missing when the sample heater is updated / changed. This will likely be solved if we are updating the heater values more continuously, but if not then we'll want that solved as well.

BrianJKoopman, 2021-08-24:
The sample heater can now be logged continuously during acquisition using the `--sample-heater` option in your ocs-site-config file or by setting the `sample_heater` value when starting the acquisition process. That was implemented in #176.

Still heater still doesn't have an option for logging continuously, but does log when changed by the Agent. Is that still a feature you'd like @kmharrington?

kmharrington, 2023-04-18:
probably? honestly I have no idea.