simonsobs/socs #47: Bluefors logs agent fails when status data is logged

JakeSpisak, 2020-02-24:
In the new, beta version of the bluefors software, the bluefors log agent throws an error when status data from the compressor, pumps, FSE, ect. is logged (see picture). 
[image]

Specifically, the error is
 `ocs-bluefors_1  | 2020-02-24T19:22:15+0000 acq:0 Crash in thread: [Failure instance: Traceback: <class 'AttributeError'>: 'NoneType' object has no attribute 'group'
ocs-bluefors_1  | /usr/lib/python3.6/threading.py:916:_bootstrap_inner
ocs-bluefors_1  | /usr/lib/python3.6/threading.py:864:run
ocs-bluefors_1  | /usr/local/lib/python3.6/dist-packages/twisted/_threads/_threadworker.py:46:work
ocs-bluefors_1  | /usr/local/lib/python3.6/dist-packages/twisted/_threads/_team.py:190:doWork
ocs-bluefors_1  | --- <exception caught here> ---
ocs-bluefors_1  | /usr/local/lib/python3.6/dist-packages/twisted/python/threadpool.py:250:inContext
ocs-bluefors_1  | /usr/local/lib/python3.6/dist-packages/twisted/python/threadpool.py:266:<lambda>
ocs-bluefors_1  | /usr/local/lib/python3.6/dist-packages/twisted/python/context.py:122:callWithContext
ocs-bluefors_1  | /usr/local/lib/python3.6/dist-packages/twisted/python/context.py:85:callWithContext
ocs-bluefors_1  | bluefors_log_tracker.py:401:start_acq
ocs-bluefors_1  | bluefors_log_tracker.py:313:read_and_publish_logs
ocs-bluefors_1  | bluefors_log_tracker.py:241:_parse_multi_value_log
ocs-bluefors_1  | ]
ocs-bluefors_1  | 2020-02-24T19:22:15+0000 acq:0 Status is now "done".`

The status log that is the source of the error is here. It is not obvious to me from the error message what the problem is. 
[Status 20-02-24.log](https://github.com/simonsobs/socs/files/4246545/Status.20-02-24.log)

BrianJKoopman, 2020-02-24:
The regex I wrote to parse the logs looks like it might have broke. Probably the format is different in the new version of the software.

BrianJKoopman, 2020-02-24:
Ah, yeah, the new logs don't have many of the fields the old log do. These are the strings in the `patterns` dictionary, here: https://github.com/simonsobs/socs/blob/c58c83597ccf0fd17eedfd86b0042dbd9f3cb957/agents/bluefors/bluefors_log_tracker.py#L125

The old logs have things like `tc400errorcode`, etc that don't exist in the new logs. Probably a quick fix to handle total misses in the matching. It'd be helpful to have a summary of new fields in the logs if that's known. Then they can be added to the known patterns.

mhasself, 2020-02-24:
Handling misses in the matching is fine... you could even go to a mode where it just transfers whatever it finds (from new_line.split(',')) into data.

But in either case: please take care not to permit the list of fields that are present to change once the Agent has started.  Changing the fields that are present in a co-sampled block violates the aggregator data model.

JakeSpisak, 2020-02-29:
Sure, new fields I can see are:
- ctrl_pres_ok
- cpatempwi
- ctr_pressure_ok

JakeSpisak, 2020-10-22:
Checking in on this since I recently installed the new beta version of the software (1.3-27). It doesn't look like the fields changed in the new version, so the latest docker build of ocs-bluefors-agent (v0.0.2-134-gb6dde01-dev) still fails when status data is logged.

BrianJKoopman, 2020-10-22:
I just added the two new fields in a PR (#114). I pushed a test image to `grumpy.physics.yale.edu/ocs-bluefors-agent:beta` if you want to give it a shot. I don't think having fields disappear mid-data collection is going to be a problem, since I believe from the toggles in the logs it either logs everything to file or nothing. There are protections in place for a block changing now though, so if you do run into that let me know (you'll see the errors in the logs.) In that case, stop the Agent, configure the Bluefors software for the logging you want, then start up the Agent.

PR #114 (2020-10-22): Add new status log fields from beta software

## Description

Adds the new fields present in status logs output by the new beta software for the DR.

## Motivation and Context

New status fields weren't in the regex patterns.

fixes #47

## How Has This Been Tested?

Still needs testing. Either build an image, or use `grumpy.physics.yale.edu/ocs-bluefors-agent:beta`.

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not w [... cut]
closed 2020-10-22 by BrianJKoopman