simonsobs/socs #49: rename the timestream aggregator

BrianJKoopman, 2020-02-27:
@mhasself points out that "timestream-aggregator" is perhaps too general/not quite accurate. We should think of a better name for this agent. Please brainstorm ideas here.

mhasself, 2020-02-27:
We're using "smurf-*" for various detector system things.  How about smurf-catcher?  I am approximately half serious.

BrianJKoopman, 2020-02-27:
I like the "smurf-" prefix idea.

Throwing out some ideas:
* "smurf-writer", since it's ultimately a G3Writer
* "smurf-stream-recorder", since it records data from the "smurf-streamer"
* "smurf-stream-writer", a combination of the two above suggestions

jlashner, 2020-03-04:
I honestly like "smurf-catcher" or "smurf-recorder". I think "smurf-stream-catcher" and "smurf-stream-recorder" are more clear, but that's getting a bit long (note that instance id's will probably also have the "c1s2" suffix for the crate and slot numbers).

BrianJKoopman, 2020-03-04:
Alright, I'll rename it to "smurf-recorder" unless there are objections. #46 and #50 should be wrapped up first.

PR #52 (2020-03-13): Rename Timestream Aggregator to Smurf Recorder
resolves #49
commit dbeb9cf8 (2020-03-13): Rename Timestream Aggregator to Smurf Recorder (#52)

Rename Timestream Aggregator to Smurf Recorder

resolves #49
closed 2020-03-13 by BrianJKoopman