simonsobs/socs #58: Add ability to turn Cryomech CPA on/off

BrianJKoopman, 2020-03-24:
There are requests for on/off control of the Cryomech CPA compressors via OCS Agent. @JakeSpisak, can you comment on the feasibility? Think we can implement this?

JakeSpisak, 2020-03-24:
Let me ping the cryomech people and see - they are the ones who gave us the python code to command the compressor.

gdevenyi, 2020-04-28:
The cryomech documentation lists the following:
> Register 40,001	‐	Enable	/	Disable	the	compressor	
> Holding	register	40001:		
> 	 	0x00FF	‐	turn	the	compressor	OFF	
>	 	0x0001	‐	Turn	the	compressor	ON	

One would need to write:
for off: 0x09 99 00 00 00 06 0000 00FF <2-byte CRC Code>
for on:  0x09 99 00 00 00 06 0000 0001 <2-byte CRC Code>
https://www.simplymodbus.ca/FC06.htm

We could either implement the CRC (https://github.com/riptideio/pymodbus/blob/master/pymodbus/utilities.py) or steal the pymodbus implementation:
https://github.com/riptideio/pymodbus/blob/master/pymodbus/utilities.py

JakeSpisak, 2020-04-28:
I actually just finished and successfully tested an implementation of this through the CRC code implementation Gabriel mentioned. I haven't gotten around to pushing things to socs yet, but maybe I can combine this with the recent cleanup draft pull request as part of a big compressor agent overhaul.

gdevenyi, 2020-04-29:
Re: CRC check, I believe the CRC data is also returned on Read, so that could also be checked, though I'm not sure its worth it since TCP is pretty robust at the other transport layers.

BrianJKoopman, 2021-08-24:
> I actually just finished and successfully tested an implementation of this through the CRC code implementation Gabriel mentioned. I haven't gotten around to pushing things to socs yet, but maybe I [...]

Hey @JakeSpisak, I just want to check in on this issue. Do you have an implementation of on/off functionality working that we could merge?