simonsobs/socs #61: MKS 390 Pressure Sensor Agent

BrianJKoopman, 2020-03-24:
Create an Agent for the MKS 390 Pressure Sensor.

JakeSpisak, 2020-03-24:
This functionality exists, and we are currently reading out this pressure sensor on this cooldown. However, it's not an agent, since there is no control all it does is output a voltage to the labjack. (Any power control would be through the iBootbar power it's plugged in to). All we did is add functionality to the labjack that allows users to input a voltage->pressure function in the site config, such that the labjack agent publishes physical values to its feed. So I'll push those labjack changes.

BrianJKoopman, 2020-03-26:
Ah, ok, so this is just a pressure sensor for the LabJack? Good to know. I'll wrap up my changes to the LabJack Agent (mostly a this point it's just adding documentation) and merge soon so you can work on that.

I'm going to close this for now, if we feel like we should re-open in the future we can do so.

closed 2020-03-26 by BrianJKoopman