simonsobs/socs #76: Possible bug reacquiring lock in 372 agent

jlashner, 2020-05-18:
I was just looking through the 372 agent usage of `release_and_acquire` and I'm confused by this behavior:
https://github.com/simonsobs/socs/blob/4c8224e2289c79ac64bb9bc0b421dbb1a4263788/agents/lakeshore372/LS372_agent.py#L202-L205

This `continue` seems a bit sketchy as on the next iteration of the loop it'll try to `release_and_acquire` a second time, but having never actually re-acquired the lock...

mhasself, 2020-05-18:
Agreed.  Not sure what the right fix is.  Probably needs to `return xxx` rather than `continue`.  One could imagine trying to reacquire again, later... but really this shouldn't be happening so failing out is probably advisable.

jlashner, 2020-05-18:
I agree, maybe return False with that warning message instead of logging it?

sanahabhimani, 2023-04-11:
Was this fixed in the agent?