simonsobs/socs #77: Failed to register basic handlers; agent probably running already

jlashner, 2020-05-27:
Currently getting this error when trying to start the pysmurf-monitor on the UCSD smurf-srv14:
```
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 Failed to register basic handlers @ observatory.pysmurf-monitor; agent probably running already.
```
I remember this happening before in daq support and I believe the solution was something like "try restarting crossbar". Has anyone else seen this recently / do we have a better solution?
Here's the full docker log:
```
cryo@smurf-srv14:~/ocs-site-configs/ucsd/smurf-srv14$ docker-compose up ocs-pysmurf-monitor
Creating ocs-pysmurf-monitor ... done
Attaching to ocs-pysmurf-monitor
ocs-pysmurf-monitor    | site_config is setting values of "udp_port" to "8200".
ocs-pysmurf-monitor    | site_config is setting values of "create_table" to "True".
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 Using OCS version 0.6.0+1.g569b428
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 ocs: starting <class 'ocs.ocs_agent.OCSAgent'> @ observatory.pysmurf-monitor
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 log_file is apparently None
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 PysmurfMonitor starting on 8200
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 Starting protocol <__main__.PysmurfMonitor object at 0x7fb29edef438>
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 transport connected
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 session joined:
ocs-pysmurf-monitor    | SessionDetails(realm=<test_realm>,
ocs-pysmurf-monitor    |                session=5918897485678291,
ocs-pysmurf-monitor    |                authid=<YCT9-RNWX-V9HJ-KMXX-MYHT-JM5T>,
ocs-pysmurf-monitor    |                authrole=<server>,
ocs-pysmurf-monitor    |                authmethod=anonymous,
ocs-pysmurf-monitor    |                authprovider=static,
ocs-pysmurf-monitor    |                authextra={'x_cb_node_id': None, 'x_cb_peer': 'tcp4:192.168.0.134:54932', 'x_cb_pid': 12},
ocs-pysmurf-monitor    |                serializer=<cbor.batched>,
ocs-pysmurf-monitor    |                resumed=None,
ocs-pysmurf-monitor    |                resumable=None,
ocs-pysmurf-monitor    |                resume_token=None)
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 Failed to register basic handlers @ observatory.pysmurf-monitor; agent probably running already.
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 session left: CloseDetails(reason=<wamp.close.normal>, message='None')
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 transport disconnected
ocs-pysmurf-monitor    | 2020-05-27T05:18:00+0000 Scheduling retry 1 to connect <twisted.internet.endpoints.TCP4ClientEndpoint object at 0x7fb29edba668> in 2.036053461502986 seconds.
```

jlashner, 2020-05-27:
Found [this stack overflow post](https://stackoverflow.com/questions/32513609/how-to-re-register-a-wamp-callee-after-a-reconnect-using-autobahn-python-with-tw) that is addressing a similar issue, and points to this page as a solution: https://crossbar.io/docs/WebSocket-Options/

I bet setting some of those production options in our crossbar config (in particular `auto_ping_interval` and `auto_ping_timeout`) might solve this problem. Might be hard to test that it's actually working though.

BrianJKoopman, 2020-05-27:
Yeah, agreed. The fact that `auto_ping_timeout` defaults to disabled makes me think it'll never drop interrupted connections. I'm sure we can come up with a way to force a connection interruption to test.

BrianJKoopman, 2020-07-02:
Have we seen this problem since making the change to the crossbar server configuration at UCSD?

jlashner, 2020-07-02:
Nope, but it happened pretty infrequently so I'm not completely convinced it's solved. We can close this though until it pops up again.

BrianJKoopman, 2020-07-02:
I'm alright with leaving it open, we should think of a way to test it.

BrianJKoopman, 2022-06-30:
Seeing as this is 2 years old and hasn't been reported recently I'm going to close this. Feel free to reopen if it comes up again.

closed 2022-06-30 by BrianJKoopman