simonsobs/socs #100: Pfeiffer Agent Block structure errors

BrianJKoopman, 2020-08-04:
I don't have time to triage this right now, but this was seen on the UCSD system during their OCS update, and I believe Penn has seen it as well:
```
2020-08-04T16:00:30+0000 Unhandled Error
Traceback (most recent call last):
  File "pfeiffer_tpg366_agent.py", line 177, in <module>
    runner.run(agent, auto_reconnect=True)
  File "/usr/local/lib/python3.6/dist-packages/autobahn/twisted/wamp.py", line 410, in run
    reactor.run()
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/base.py", line 1283, in run
    self.mainLoop()
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/base.py", line 1292, in mainLoop
    self.runUntilCurrent()
--- <exception caught here> ---
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/base.py", line 886, in runUntilCurrent
    f(*a, **kw)
  File "/app/ocs/ocs/ocs_feed.py", line 208, in publish_message
    b.append(message)
  File "/app/ocs/ocs/ocs_feed.py", line 34, in append
    raise Exception("Block structure does not match: {}".format(self.name))
builtins.Exception: Block structure does not match: pressures
```

SOCS Version (and Docker tag): v0.0.2-67-g71a8640
OCS Version: 0.6.0+1.g569b428

BrianJKoopman, 2020-08-04:
This might be proceeded by a timeout, could be the hardware losing its network connection for some reason. I got the agent back online at UCSD after restarting the container, but it very quickly ran into:
```
2020-08-04T20:27:06+0000 acq:0 Crash in thread: [Failure instance: Traceback: <class 'socket.timeout'>: timed out
/usr/lib/python3.6/threading.py:916:_bootstrap_inner                                      
/usr/lib/python3.6/threading.py:864:run
/usr/local/lib/python3.6/dist-packages/twisted/_threads/_threadworker.py:46:work           
/usr/local/lib/python3.6/dist-packages/twisted/_threads/_team.py:190:doWork
--- <exception caught here> ---                                                                 
/usr/local/lib/python3.6/dist-packages/twisted/python/threadpool.py:250:inContext
/usr/local/lib/python3.6/dist-packages/twisted/python/threadpool.py:266:<lambda>
/usr/local/lib/python3.6/dist-packages/twisted/python/context.py:122:callWithContext                  
/usr/local/lib/python3.6/dist-packages/twisted/python/context.py:85:callWithContext
pfeiffer_tpg366_agent.py:139:start_acq                         
pfeiffer_tpg366_agent.py:73:read_pressure_all
]
```

Subsequent restarts led to the initial Block structure error.

tanaybhandarkar, 2021-02-18:
I updated the Pfeiffer agent code and pushed it to the Pfeiffer branch. I believe this error was triggered when the agent tried to query a pressure while channels were off. The agent now checks if a channel is off, and reports the pressure as 0 if the channel is actually off.

closed 2024-11-22 by BrianJKoopman