simonsobs/socs #106: DSE (Generator) monitoring

mhasself, 2020-09-01:
An Agent is needed to collect generator stats from the Deep Sea Electronics interfaces. @DaveBoettger suggests to do this over modbus, using pyModbusTCP.  This has twisted compatibility, e.g. see https://pymodbus.readthedocs.io/en/v2.3.0/source/example/asynchronous_processor.html .

For safety, the Agent should not implement any control (in contrast to monitoring) functionality.

retrodaredevil, 2021-05-22:
@mhasself this is one of the few places on GitHub I could find a mention of DSE Generator monitoring. I was going to be doing this soon and I was wondering if you had any documentation on protocols used by Deep Sea Electronics Generator Control modules. Specifically a DSE3110. Any links would be helpful. I've only been doing research on this for about an hour at the time of writing this, so maybe I'll find it later.

mhasself, 2021-05-22:
@retrodaredevil Sorry, I don't know to much about this specific hardware.  Someone in our group will eventually implement code for this purpose but I don't have a timeline and I don't know much, personally, about the hardware.

Some quick looking through datasheets suggests that the details for using modbus for some particular device need to be requestsed by email from Deepsea.  (E.g. "CONNECTING USING A THIRD PARTY MODBUS MASTER" in the software manual for DSE857.)

retrodaredevil, 2023-06-23:
Hi @davidvng. I've been following this issue for over 2 years so I was happy to see this feature being complete today, but I'm not sure if it's something I can use so I had some questions. (I assume you have the most knowledge on this as it looks like you implemented the bulk of the PR, but anyone else also feel free to answer).

I have a DSE3110 that is CAN enabled. I haven't played with it yet, so I'm really still in the research phase (as I was 2 years ago). I was wondering if you could point me in the right direction for finding documentation for what I am trying to accomplish. I was also wondering if you had any idea if your agent can be used to monitor a CAN enabled DSE3110 or if it is just for communicating serially. I would assume that the CAN protocol would not use Modbus, but I really have no idea how it works and was wondering if you could help me understand it better.

davidvng, 2023-06-23:
Hi @retrodaredevil, we are using the DSE8610 MKII which is compatible with Modbus. From initial searches for your model, DSE3110, it doesn't seem to support Modbus. We don't have plans to expand our agent to use the CAN protocol. I believe you could modify the agent to use something like the `python-can` python package instead of `pyModbusTCP`. For more documentation on your specific model I suggest going to the Deep Sea Electronics website, making an account, and requesting the manuals.

PR #436 (2023-06-22): Generator Agent

## Description

Created new agent to monitor on-site generators. A standard acq process talks to the generator controller via Modbus and queries registers.

## Motivation and Context

New agent. Resolves #106.

## How Has This Been Tested?

Tested agent with on-site generator: able to connect and query, and verified connection drops don't crash the agent.
Requires module `pyModbusTCP`

## Types of changes

- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would ca [... cut]