simonsobs/socs #112: Smurf-recorder setting stream-id in path to None

jlashner, 2020-10-10:
Sometimes the smurf-recorder is setting the stream-id component in the data path to None even if there is a stream-id present in the frames. I think it does this when it receives the PipelineInfo frame at the start, since that doesn't have the `sostream_id` key. 

For example, on the Penn system the file /data/timestreams/16022/None/1602279295_000.g3 contains the following frames:
```
Frame (PipelineInfo) [
"08-Oct-2020:17:18:10.069906000" (spt3g.core.G3PipelineInfo) => master branch, no local diffs
]
Frame (Observation) [
"frame_num" (spt3g.core.G3Int) => 0
"session_id" (spt3g.core.G3Int) => 1602187012
"sostream_id" (spt3g.core.G3String) => "crate1slot2"
"sostream_version" (spt3g.core.G3Int) => 1
"time" (spt3g.core.G3Time) => 08-Oct-2020:19:56:52.054788000
]
Frame (Wiring) [
"dump" (spt3g.core.G3Int) => 0
"frame_num" (spt3g.core.G3Int) => 170
"session_id" (spt3g.core.G3Int) => 1602187012
"sostream_id" (spt3g.core.G3String) => "crate1slot2"
"sostream_version" (spt3g.core.G3Int) => 1
"status" (spt3g.core.G3String) => "AMCc.SmurfProcessor.ChannelMapper.NumChannels: 63
AMCc.SmurfProcessor.PostDataEmulator.Offset: 0
AMCc.SmurfProcessor.PostDataEmulator.Amplitude: 4294967295
"
"time" (spt3g.core.G3Time) => 08-Oct-2020:19:57:27.892115000
]
```

jlashner, 2020-10-11:
I'm tempted to just discard the G3PipelineInfo frame here which I think would solve the problem: https://github.com/simonsobs/socs/blob/5324b0768526bf427fdf5c13d7932f3551ef2f83/socs/agent/smurf_recorder.py#L263

Does anyone think we need to be saving the pipeline info frame?

mhasself, 2020-10-12:
Fine by me.

BrianJKoopman, 2020-10-16:
Are you setting a default stream_id with the `--stream-id` argument to the Agent? That argument is used if there isn't one in the frames. That should at least prevent you from ending up with "None" in this case.

PR #134 (2021-02-01): Add some quality of life improvements to smurf-rec
## Description
Changes consist of:
1. Check if socket is open before trying to create G3Streamer to
   minimize log_fatals
2. Change some info statents into debug, and some debug statements to
   info
3. Discards G3PipelineInfo frames to fix stream-id-less file on startup
   bug.

## Motivation and Context

This fixes some small bugs in the smurf-recorder and makes it easier to monitor during operation. Mainly did this in order to further debug issue #78, but this should also resolve #112.

## How Has This Been Tested?

Tested at UCSD by hamme [... cut]
closed 2021-02-01 by jlashner