simonsobs/socs #119: Pysmurf monitor crashes if file does not exist

jlashner, 2020-11-05:
Pysmurf monitor crashes if a registered files does not exist. We should check for this, and probably try to restart the process on failure. 

Error logs from k2so's smurf-server:
```
2020-11-05T23:17:30+0000 New file: /data/smurf_data/20201105/1604617842/outputs/1604617875_iv_bias_all
2020-11-05T23:17:30+0000 Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python3.6/dist-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python3.6/dist-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
--- <exception caught here> ---
  File "/usr/local/lib/python3.6/dist-packages/twisted/internet/udp.py", line 249, in doRead
    self.protocol.datagramReceived(data, addr)
  File "pysmurf_monitor.py", line 124, in datagramReceived
    'md5sum':               get_md5sum(d['path']),
  File "/app/socs/socs/util.py", line 14, in get_md5sum
    for line in open(filename, 'rb'):
builtins.FileNotFoundError: [Errno 2] No such file or directory: '/data/smurf_data/20201105/1604617842/outputs/1604617875_iv_bias_all'
2020-11-05T23:17:30+0000 New file: /data/smurf_data/20201105/1604617842/outputs/1604617875_iv_raw_data.npy
2020-11-05T23:17:30+0000 session left: CloseDetails(reason=<wamp.close.transport_lost>, message='WAMP transport was lost without closing the session 3862389767194455 before')
2020-11-05T23:17:30+0000 transport disconnected
2020-11-05T23:17:30+0000 Scheduling retry 1 to connect <twisted.internet.endpoints.TCP4ClientEndpoint object at 0x7ff7ce1d8668> in 2.1734791071380224 seconds.
2020-11-05T23:17:30+0000 (UDP Port 8200 Closed)
2020-11-05T23:17:30+0000 Stopping protocol <__main__.PysmurfMonitor object at 0x7ff7ce20d438>
2020-11-05T23:17:30+0000 Main loop terminated.
```

BrianJKoopman, 2020-11-06:
I agree we should catch that, however I'm curious why the file doesn't exist. Is the monitor being informed of a file before it's available for the md5sum to be computed?

jlashner, 2020-11-06:
After looking into it, seems like the `FileNotFoundError` and the disconnect are separate problems, and in most cases the monitor continues to run even if the file is not found.

Also turns out the reason the file doesn't exist is that I forgot to set the format as `npy` in the pysmurf publish statement:
```
        basename, _ = os.path.splitext(os.path.basename(datafile))
        path = os.path.join(self.output_dir, basename + '_iv_bias_all')
        np.save(path, bias)

        # publisher announcement
        self.pub.register_file(path, 'iv_bias')
```
so the file is actually being saved with the `.npy` extension even though it's not in the path name. The publisher checks for this specific case but only if it's published with `format='npy'`

BrianJKoopman, 2021-05-19:
Just following up on this. Was this fixed upstream yet?

BrianJKoopman, 2022-05-27:
It's been a year, this likely isn't an issue anymore, closing as stale. Feel free to reopen if it is.

closed 2022-05-27 by BrianJKoopman