simonsobs/socs #140: Pysmurf / sodetlib functions lead to a lot of "not real" .g3 data files.

kmharrington, 2021-02-15:
I think the errors are probably in code here? Not quite sure.  I had previously noticed a lot of streamed data from the k2so system where the `.g3` files had data fields that were (1 x nsamps). These were times where there wasn't meant to be any data functions. 

I just found it again in Chicago's system, but this time I know what we were doing. This morning we ran:

```
cfg = DetConfig()
cfg.load_config_files(slot=2)
 
S = cfg.get_smurf_control(dump_configs=True, make_logfile=False)
S.setup()

S.set_att_dc(band, 0)
S.set_att_uc(band, 30)
 
S.load_tune()
print(S.tune_file)
 
[ 2021-02-15 17:07:27 ]  Defaulting to last tuning: /data/smurf_data/tune/1613174193_tune.npy
[ 2021-02-15 17:07:27 ]  Done loading tuning
[ 2021-02-15 17:07:27 ]  Loading tune data for all bands=[0, 1, 2, 3].
/data/smurf_data/tune/1613174193_tune.npy
 
S.relock(band)
 
for _ in range(3):
    S.run_serial_gradient_descent(band)
    S.run_serial_eta_scan(band)

f, df, sync = S.tracking_setup(band, reset_rate_khz=4,lms_freq_hz=None,meas_lms_freq=True,
                 fraction_full_scale=0.2,
                 make_plot=True,show_plot=True,channel=S.which_on(band),nsamp=2**18,
                 feedback_start_frac=0.02,feedback_end_frac=0.94)
```

This resulted in these folders in our smurf directory:
```
kmharrin@simons1:/mnt/so1/data/chicago-latrt/smurf/16134/crate1slot2$ ls
1613407584_config          1613409844_tracking_setup  
```
Somehow we started "streaming" `.g3` timestreams with data that are (1 x Nsamps) at ctime=1613407652. This looks like it was after we ran the configuration startup and before we ran tracking setup.

jlashner, 2021-02-15:
I think this is a pysmurf issue. These lines enable streaming for one channel in the setup function: https://github.com/slaclab/pysmurf/blob/d55dfeba7345f28d610328ad1e447ebe23799ffb/python/pysmurf/client/base/smurf_control.py#L609-L613
so I think this behavior is actually correct on our end. We should try to figure out why they're doing this and if it is a serious problem, disable it.

BrianJKoopman, 2021-09-30:
What's the status of this? Was it a pysmurf issue? Can we close this here?

jlashner, 2021-10-06:
This was solved by the most recent version of the smurf-streamer, which now triggers on the OpenFile register instead of enableStreaming.