simonsobs/socs #166: Influx Alert Insanity

jlashner, 2021-05-18:
At many institutions now we've seen grafana alerts continuously trigger with NoData errors. I believe we first noticed it in K2SO when we started doing beam maps like 3 or 4 months ago. However, we have since noticed it at many other institutions like princeton, u-chicago, and most recently with SAT1 thermometry alarms. This problem generally looks like a mass triggering of many alarms with NoData warnings, and then the alerts being shown as resolved a few minutes later after it receives data again.

Has there been a change in the last few months or so that could possibly be causing this? Or is it an unexpected result of labs scaling up? I was suspecting it might be related to labs starting to readout UFM's which use more memory and CPU, but with the recent SAT1 alarms I'm not so sure. 

Anyway, this seems like a critical issue since many labs are relying on influx live monitoring tools.

BrianJKoopman, 2021-05-18:
The SAT1 instance we found related to the configuration of the 372 changing. The scan duration increased from just a few seconds to ~90 seconds. The 372 was set to autoscan, and was scanning over six sensors. This caused the alarm, which inspects the last 5 minutes of data, to go off regularly, as it would take about 7.5 minutes to scan over the channels between a sensor checked by the alarm.

The configuration change seems to coincide with a power cycle, which is strange, but emphasizes the need for https://github.com/simonsobs/socs/issues/96.

For other instances where alarms unexpectedly have no data it'd be good to get the following information:
* Alarm configuration
  * What's being checked and how often -- essentially the "conditions" section of the alert
* Computer hardware (especially relevant is the storage medium being used to store in the InfluxDB)
* List of running Agents (raw output from `docker ps` at the time of the alarms is ideal)
* Any recent changes to the system

BrianJKoopman, 2022-04-15:
As far as I know this issues hasn't been seen in quite a while. Closing. Feel free to re-open if it's still a problem.

closed 2022-04-15 by BrianJKoopman