simonsobs/socs #185: Bluefors agent crashing with Bluefors PC windows directory mounted

swh76, 2021-08-05:
I've been running into this issue sporadically ; the BlueforsAgent is crashing with this error;

```
2021-08-03T02:02:10+0000 The Bluefors Agent cannot process error logs. Not publishing.
2021-08-03T11:36:55+0000 The Bluefors Agent cannot process error logs. Not publishing.
2021-08-03T11:36:55+0000 The Bluefors Agent cannot process error logs. Not publishing.
2021-08-03T13:00:17+0000 acq:0 Crash in thread: [Failure instance: Traceback: <class 'ValueError'>: not enough values to unpack (expected 3, got 1)
/usr/lib/python3.8/threading.py:932:_bootstrap_inner
/usr/lib/python3.8/threading.py:870:run
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_threadworker.py:47:work
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_team.py:181:doWork
--- <exception caught here> ---
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:238:inContext
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:254:<lambda>
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:118:callWithContext
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:83:callWithContext
bluefors_log_tracker.py:488:start_acq
bluefors_log_tracker.py:385:read_and_publish_logs
bluefors_log_tracker.py:227:_parse_single_value_log
]
2021-08-03T13:00:17+0000 acq:0 Status is now "done".
```

I'm running the OCS dockers on a linux server and mounting the windows Bluefors log directory on that server, so that may be playing a role.  Zach Huber at Cornell who is also running the Bluefors Agent this way (with the windows log directory mounted on linux) reports similar crashes.  Restarting the agent seems to get things going again.  Thanks!  Shawn

BrianJKoopman, 2022-01-24:
I introduced some exception handling for this in a test image, which then caught the following on Shawn's system:
```
2022-01-20T00:00:37+0000 /logs//22-01-20/Status_22-01-20.log not yet open, opening...
2022-01-20T00:57:45+0000 Could not unpack values from line: ""
2022-01-20T00:57:45+0000 Unable to parse log line. Skipping this line: 
2022-01-20T00:57:45+0000 Invalid line of type "maxiguage" from log "maxigauge"
2022-01-20T00:57:45+0000 time data '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' does not match format '%d-%m-%y,%H:%M:%S'
2022-01-20T00:57:45+0000 Unable to parse log line. Skipping this line: 
2022-01-20T00:57:45+0000 Invalid line of type "status" from log "status"
2022-01-20T00:57:45+0000 not enough values to unpack (expected at least 2, got 1)
2022-01-20T13:00:13+0000 Could not unpack values from line: ""
```
Going to figure out why we're getting all that hex code occasionally, and will PR the exception handling soon.

BrianJKoopman, 2022-01-27:
Another error to handle has come up:
```
$ docker logs ea818514961d
2022-01-24T00:00:00+0000 Closed file: /logs//22-01-23/maxigauge 22-01-23.log
2022-01-24T00:00:00+0000 Closed file: /logs//22-01-23/Status_22-01-23.log
2022-01-24T00:00:00+0000 Closed file: /logs//22-01-23/Flowmeter 22-01-23.log
2022-01-24T00:00:13+0000 /logs//22-01-24/Flowmeter 22-01-24.log not yet open, opening...
2022-01-24T00:00:45+0000 /logs//22-01-24/maxigauge 22-01-24.log not yet open, opening...
2022-01-24T00:00:59+0000 /logs//22-01-24/Status_22-01-24.log not yet open, opening...
2022-01-25T00:00:00+0000 Closed file: /logs//22-01-24/Flowmeter 22-01-24.log
2022-01-25T00:00:00+0000 Closed file: /logs//22-01-24/maxigauge 22-01-24.log
2022-01-25T00:00:00+0000 Closed file: /logs//22-01-24/Status_22-01-24.log
2022-01-25T00:00:41+0000 /logs//22-01-25/Flowmeter 22-01-25.log not yet open, opening...
2022-01-25T00:00:53+0000 /logs//22-01-25/Status_22-01-25.log not yet open, opening...
2022-01-25T00:01:01+0000 /logs//22-01-25/maxigauge 22-01-25.log not yet open, opening...
2022-01-25T14:36:45+0000 acq:0 CRASH: [Failure instance: Traceback: <class 'OSError'>: [Errno 112] Host is down: '/logs//22-01-25/Flowmeter 22-01-25.log'
/usr/lib/python3.8/threading.py:932:_bootstrap_inner
/usr/lib/python3.8/threading.py:870:run
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_threadworker.py:47:work
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_team.py:181:doWork
--- <exception caught here> ---
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:238:inContext
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:254:<lambda>
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:118:callWithContext
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:83:callWithContext
bluefors_log_tracker.py:529:acq
bluefors_log_tracker.py:410:read_and_publish_logs
]
2022-01-25T14:36:45+0000 acq:0 Status is now "done".
```

zhuber21, 2022-11-28:
Another error that we just saw on our similar system (Bluefors logs to lnx201 (a Linux computer) and the agent pulls from there): 

```
2022-11-24T00:00:24+0000 /logs/22-11-24/maxigauge 22-11-24.log not yet open, opening...
2022-11-24T00:01:06+0000 /logs/22-11-24/Flowmeter 22-11-24.log not yet open, opening...
2022-11-25T00:00:00+0000 Closed file: /logs/22-11-24/maxigauge 22-11-24.log
2022-11-25T00:00:00+0000 Closed file: /logs/22-11-24/Flowmeter 22-11-24.log
2022-11-25T00:01:15+0000 /logs/22-11-25/maxigauge 22-11-25.log not yet open, opening...
2022-11-25T00:01:15+0000 /logs/22-11-25/Flowmeter 22-11-25.log not yet open, opening...
2022-11-26T00:00:00+0000 Closed file: /logs/22-11-25/maxigauge 22-11-25.log
2022-11-26T00:00:00+0000 Closed file: /logs/22-11-25/Flowmeter 22-11-25.log
2022-11-26T00:00:12+0000 /logs/22-11-26/maxigauge 22-11-26.log not yet open, opening...
2022-11-26T00:01:15+0000 /logs/22-11-26/Flowmeter 22-11-26.log not yet open, opening...
2022-11-27T00:00:00+0000 Closed file: /logs/22-11-26/maxigauge 22-11-26.log
2022-11-27T00:00:00+0000 Closed file: /logs/22-11-26/Flowmeter 22-11-26.log
2022-11-27T00:01:12+0000 /logs/22-11-27/Flowmeter 22-11-27.log not yet open, opening...
2022-11-27T00:01:12+0000 /logs/22-11-27/maxigauge 22-11-27.log not yet open, opening...
2022-11-27T13:09:13+0000 acq:0 CRASH: [Failure instance: Traceback: <class 'OSError'>: [Errno 5] Input/output error: '/logs/22-11-27/Flowmeter 22-11-27.log'
/usr/lib/python3.8/threading.py:932:_bootstrap_inner
/usr/lib/python3.8/threading.py:870:run
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_threadworker.py:47:work
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_team.py:182:doWork
--- <exception caught here> ---
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:244:inContext
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:260:<lambda>
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:117:callWithContext
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:82:callWithContext
/usr/local/lib/python3.8/dist-packages/socs/agents/bluefors/agent.py:497:acq
/usr/local/lib/python3.8/dist-packages/socs/agents/bluefors/agent.py:380:read_and_publish_logs
/usr/local/lib/python3.8/dist-packages/socs/agents/bluefors/agent.py:125:reopen_file
]
2022-11-27T13:09:13+0000 acq:0 Status is now "done".
```

zhuber21, 2022-11-28:
It looks like our mount to the samba drive (lnx201) had been broken for some reason, which is possibly related to this error. I have to remake the samba connection every so often and haven't really figured out why.