simonsobs/socs #259: Add "get_X" and "get_state" functions to LS372 agent

msilvafe, 2022-02-26:
Currently there are only a few tasks exposed for polling the current lakeshore settings:
- `get_dwell`
- `get_excitation`
- `get_resistance_range`
- `get_still_output`
The other tasks are for setting things only i.e. `set_active_channel`, `set_heater_range`, `set_pid`, `set_output_mode` and don't have equivalent `get` functions. It would be very helpful to be able to poll the current state of the lakeshore by either having equivalent `get_X` functions for all `set_X` functions and/or a `get_state` function that returns a dictionary or similar output with all of the current settings to be able to tell what state you are in when walking up to the system blind.

BrianJKoopman, 2022-06-01:
After discussion on today's dev call, I agree that a "get_all_settings" task would be good. Depending on how thorough it is it might be a little slow though.

dpdutcher, 2022-06-22:
bump.
I especially would like `get_channel_attribute` and `get_heater_attribute`, ala the LS370 agent, and the ability to change the PID channel.