simonsobs/socs #348: Support Keithley 2260B series PSU

BrianJKoopman, 2022-08-29:
Penn has a new model of PSU, a [Keithley 2260B series](https://www.tek.com/en/support/product-support?series=Keithley%202260B%20Series&type=manual). The SCPI PSU Agent needs to be updated to handle direct communication via Ethernet, rather than over the GPIB to Ethernet converter.

raymondllata, 2023-08-16:
Hi! I'm Raymond Llata a research intern @SLAC Laboratory right now. I actually just finished modifying the PSU agent to work with the Keysight E36312A, which also only communicates over Ethernet. I would love to contribute this to SOCS but I had two questions: 
1) I can't open this SO Developer Guide so (https://simons1.princeton.edu/docs/so_dev_guide/) I'm unsure what info it contains.
2) I made my agent separate from the original PSU agent, since it would not use the provided PSU/Prologix Interface's. It is imperative two have the single agent be able to detect what interface is being used and respond accordingly?

newburgh, 2023-08-16:
@BrianJKoopman It is a good point if our dev guide isn't public right now. 

For one vs two agents: It is not imperative they're the same, I think its a good idea to have your interface in a separate agent and then combine if it is clear that is straightforward. Perhaps naming the agent appropriately would be wise (KeithleyPSU_TCP or whatever you're using).

raymondllata, 2023-08-16:
I submitted a configuration where it will use the default interface if both the IP and GPIB configurations are supplied. Alternatively, if only IP is provided it will connect via an IP and a Port. Right now, I have the port set to 5025 (as far as I am aware this is default for TCP protocol). Would it be better to allow for an Port argument as well (presumably if a different port is required?)

BrianJKoopman, 2026-03-04:
Support for direct Ethernet connection was added in #715, so this might work now, even though the 2260B model isn't listed in our officially supported models. I haven't heard about this PSU being used for a while, so I'm going to close this issue, but feel free to reopen if this is still needed.