simonsobs/socs #379: Registry agent exits out due to blockstructure mismatch

jlashner, 2022-12-15:
This is likely caused by an agent being restarted / updated with additional operations. Registry needs to be resiliant to this. Maybe changing the structure so each operation instead of each agent has it's own block.

```
2022-12-14T20:39:36+0000 OCSAgent.onUserError(): "While firing <bound method Registry._register_heartbeat of <ocs.agents.registry.agent.Registry object at 0x7f4f02f3a4f0>> subscribed under 5302732399893542."
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/autobahn/twisted/websocket.py", line 387, in _onMessage
    self.onMessage(payload, isBinary)
  File "/usr/local/lib/python3.8/dist-packages/autobahn/wamp/websocket.py", line 106, in onMessage
    self._session.onMessage(msg)
  File "/usr/local/lib/python3.8/dist-packages/autobahn/wamp/protocol.py", line 791, in onMessage
    future = txaio.as_future(handler.fn, *invoke_args, **invoke_kwargs)
  File "/usr/local/lib/python3.8/dist-packages/txaio/tx.py", line 369, in as_future
    return maybeDeferred(fun, *args, **kwargs)
--- <exception caught here> ---
  File "/usr/local/lib/python3.8/dist-packages/twisted/internet/defer.py", line 206, in maybeDeferred
    result = f(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/ocs/agents/registry/agent.py", line 122, in _register_heartbeat
    self._publish_agent_ops(reg_agent)
  File "/usr/local/lib/python3.8/dist-packages/ocs/agents/registry/agent.py", line 142, in _publish_agent_ops
    self.agent.publish_to_feed('agent_operations', msg)
  File "/usr/local/lib/python3.8/dist-packages/ocs/ocs_agent.py", line 540, in publish_to_feed
    self.feeds[feed_name].publish_message(message)
  File "/usr/local/lib/python3.8/dist-packages/ocs/ocs_feed.py", line 224, in publish_message
    b.append(message)
  File "/usr/local/lib/python3.8/dist-packages/ocs/ocs_feed.py", line 36, in append
    raise Exception("Block structure does not match: {}".format(self.name))
builtins.Exception: Block structure does not match: observatory.pysmurf-controller-c2s2
```

jlashner, 2022-12-15:
Oops I realized this should be an OCS issue

BrianJKoopman, 2022-12-16:
Totally, and I see you re-opened over there. Going to close this one. (Issues can be transferred between repos, btw. "Transfer issue" in the lower right.)

closed 2022-12-16 by BrianJKoopman