simonsobs/socs #381: Regular reporting of LS372 heater values

jlashner, 2022-12-16:
Currently it seems that the LS372 agents report heater values on change, but it would be nice to have these be regularly queried and reported during the acquisition process.

BrianJKoopman, 2022-12-16:
Which heater? The sample heater?

BrianJKoopman, 2022-12-16:
The sample heater is published regularly in the `acq` process if `--sample-heater` is set in the SCF. The still heater doesn't have a similar functionality -- it's typically set to a value and kept static.

jlashner, 2022-12-16:
ah ok I will add that to the SAT cfg files. I think we should add regular reporting for the still also. If we're just setting it once and forgetting about it, then it will not appear in grafana if the time range your querying doesn't include the time at which the heater was set, which can be an issue with understanding the current state of the cryostat.

BrianJKoopman, 2022-12-16:
Ah, yeah, great point, we should add the still heater to the main acq loop as well, either with a new arg, or perhaps `--sample-heater` should just be `--heaters`.

The reason I made it optional was because acq used to be slowed down by more queries, and we were already at only ~4 Hz. I do think it's been sped up to ~10 Hz or better, so probably we should try to see if adding these outright without the flag affects the sample rate, and if not, then just do that.

jlashner, 2022-12-16:
Ya that makes sense to me. Maybe we can make this default to True, and change it so that it only queries heater values like once every few seconds. I don't think we need this as the same rate as other thermometry measurements.

BrianJKoopman, 2022-12-16:
I think that's true for the still heater, but the sample heater is probably being used to servo and changes quite rapidly. I think it'd still be good to query that at whatever cadence the `acq` loop is running.

jlashner, 2022-12-16:
Do you mean in the custom PID loop? that's fine with me, but if it ends up being an issue we may be able to report the commanded heater values from the PID computation without querying the lakeshore

BrianJKoopman, 2022-12-20:
I meant from the PID provided by the device itself. I believe only UCSD is using that custom software PID loop (especially since it's a local image on their system.)

BrianJKoopman, 2023-10-11:
Just bumping this, we should work on enabling consistent readout of the Still heater at some regular cadence, though probably not every loop in `acq`.