simonsobs/socs #392: Fix Integral Term Bug in Custom 372 PID

mjrand, 2023-01-23:
The custom 372 PID is slow to turn on the heater after the sample temperature is greater than the setpoint for long periods of time. I suspect that the Integral contribution term in the PID is invisibly becoming very negative so that the heater cannot be turned back on immediately when the sample temperature falls below the setpoint. This can delay the PID from turning on the heater by a time greater than the temperature was above the setpoint. There is no issue when the sample temperature is greater than the setpoint for very brief periods during normal PID'ing.

I suspect the issue is slightly more complicated than just enforcing the integral term to always be positive. This is mostly an issue because there's no such thing as a negative heater %. I'd suspect the solution is something like pausing any updating to the integral term if the heater value ever becomes 0.

BrianJKoopman, 2024-09-16:
@mjrand is this fully resolved by #753?

mjrand, 2024-09-16:
Yes and I implemented a solution of just preventing I from becoming negative even though I said that wouldn't work (lol)

BrianJKoopman, 2024-09-16:
> Yes and I implemented a solution of just preventing I from becoming negative even though I said that wouldn't work (lol)

Heh, that's why I asked. Closing as resolved by #753.

closed 2024-09-16 by BrianJKoopman