simonsobs/socs #417: ibootbar -- lock some outlets

mhasself, 2023-02-23:
Can we add an option to annotate outlets in the ibootbar/synaccess agent so that they can be treated slightly differently (especially by ocs-web)?  The goal is to make it a bit difficult for someone to power off critical network infrastructure remotely (leading to an unrecoverable outage).  This could be a passive notation (the Agent simply provides information about what outlets are critical to network) or explicit prevention (the Agent provides that information but also rejects state change requests on those outlets) -- for ocs-web it's pretty much the same.

I'm imagining either a config file or simply cmdline args like "--lock-outlet login2-1 --lock-outlet switch3".  The config file might be desirable in the longer term, when we start having "access levels" and the ability to escalate privileges.  But cmdline args would be great for now.

davidvng, 2023-03-30:
Should there then also be a way to unlock an outlet if needed? If there is only a lock outlet option I think the only way to unlock is if the agent is re-initialized?

mhasself, 2023-03-31:
Ya, I think that's a reasonable thing to build in.  One solution might be to have a task that lets you lock/unlock outlets.

When "Access Levels" are implemented, that task could be restricted to more privileged users.

PR #446 (2023-06-06): Updates to ibootbar Agent

## Description

Various updates to the ibootbar agent. Includes logic change to handling connection drops, which is the same with UPS agent. Created new task `lock_outlet` that can lock/unlock a specified outlet. If an outlet is locked, other tasks such as `set_outlet` will fail. Agent can be initialized with locked outlets using args. For example, `["--lock_outlet", [1, 2, 3]]` in `default.yaml` or `--lock-outlet 1 2 3` on command line will startup agent with outlets 1, 2, and 3 locked.

## Motivation and Context

Resolves #417, resolves #437.

## How Has This Been  [... cut]
closed 2023-06-06 by BrianJKoopman