simonsobs/socs #418: Serial collisions cause LS240 to respond with garbage and crashes the agent

BrianJKoopman, 2023-02-23:
If multiple programs try to communicate with the 240, junk might be sent back when the agent is querying the sensors. This junk then can't be cast to float here:
https://github.com/simonsobs/socs/blob/9ab5c5e08ab328bd94c421399a6b5d3288518547/socs/Lakeshore/Lakeshore240.py#L337

This causes the `acq` process to crash. We should catch these instances and log the error, but just move onto the next iteration of the loop and continue querying.

```
2023-02-22T23:53:34+0000 startup-op: launching init_lakeshore
2023-02-22T23:53:34+0000 start called for init_lakeshore
2023-02-22T23:53:34+0000 init_lakeshore:0 Status is now "starting".
2023-02-22T23:53:34+0000 init_lakeshore:0 Status is now "starting".
2023-02-22T23:53:34+0000 attempt 0
2023-02-22T23:53:34+0000 Initialized Lakeshore module: Lake Shore Model 240-2P          (LSA21OT)
2023-02-22T23:53:34+0000 start called for acq
2023-02-22T23:53:34+0000 init_lakeshore:0 Lakeshore initialized with ID: LSA21OT
2023-02-22T23:53:34+0000 acq:1 Status is now "starting".
2023-02-22T23:53:34+0000 init_lakeshore:0 Lakeshore module initialized.
2023-02-22T23:53:34+0000 init_lakeshore:0 Status is now "done".
2023-02-22T23:53:34+0000 acq:1 Status is now "running".
2023-02-23T05:34:52+0000 acq:1 CRASH: [Failure instance: Traceback: <class 'ValueError'>: could not convert string to float: '0292.4\r+00.5789'
/usr/lib/python3.8/threading.py:932:_bootstrap_inner
/usr/lib/python3.8/threading.py:870:run
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_threadworker.py:47:work
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_team.py:182:doWork
--- <exception caught here> ---
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:244:inContext
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:260:<lambda>
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:117:callWithContext
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:82:callWithContext
/usr/local/lib/python3.8/dist-packages/socs/agents/lakeshore240/agent.py:221:acq
/usr/local/lib/python3.8/dist-packages/socs/Lakeshore/Lakeshore240.py:337:get_reading
]
```

```
2023-02-22T23:00:16+0000 startup-op: launching init_lakeshore
2023-02-22T23:00:16+0000 start called for init_lakeshore
2023-02-22T23:00:16+0000 init_lakeshore:0 Status is now "starting".
2023-02-22T23:00:16+0000 init_lakeshore:0 Status is now "starting".
2023-02-22T23:00:16+0000 attempt 0
2023-02-22T23:00:16+0000 Initialized Lakeshore module: Lake Shore Model 240-2P          (LSA21OT)
2023-02-22T23:00:16+0000 start called for acq
2023-02-22T23:00:16+0000 init_lakeshore:0 Lakeshore initialized with ID: LSA21OT
2023-02-22T23:00:16+0000 acq:1 Status is now "starting".
2023-02-22T23:00:16+0000 init_lakeshore:0 Lakeshore module initialized.
2023-02-22T23:00:16+0000 init_lakeshore:0 Status is now "done".
2023-02-22T23:00:16+0000 acq:1 Status is now "running".
2023-02-22T23:06:57+0000 acq:1 CRASH: [Failure instance: Traceback: <class 'ValueError'>: could not convert string to float: 'D-670S-.L,6SN  235000,\r+0.00000'
/usr/lib/python3.8/threading.py:932:_bootstrap_inner
/usr/lib/python3.8/threading.py:870:run
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_threadworker.py:47:work
/usr/local/lib/python3.8/dist-packages/twisted/_threads/_team.py:182:doWork
--- <exception caught here> ---
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:244:inContext
/usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:260:<lambda>
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:117:callWithContext
/usr/local/lib/python3.8/dist-packages/twisted/python/context.py:82:callWithContext
/usr/local/lib/python3.8/dist-packages/socs/agents/lakeshore240/agent.py:221:acq
/usr/local/lib/python3.8/dist-packages/socs/Lakeshore/Lakeshore240.py:337:get_reading
]
2023-02-22T23:06:57+0000 acq:1 Status is now "done".
2023-02-22T23:53:13+0000 Received SIGTERM, shutting down.
```

davidvng, 2023-03-30:
This error looks exactly like a connection error I'm getting while testing connection drops (unplugging the USB). In my fix, I am able to address this by moving onto the next iteration without the process crashing. For documentation purposes, I label this error as a connection error, but I can add something like: "This might also be multiple programs trying to communicate with the device."

BrianJKoopman, 2023-04-04:
Interesting. I do like that additional log message you suggest.

I'm catching up on emails/PRs, and hope to get to #430 soon.

closed 2026-03-09 by BrianJKoopman