simonsobs/socs #455: PMX voltage and current flag monitoring

jlashner, 2023-06-07:
We need functionality in the PMX agent that allows us to monitor whether or not the voltage and current limit flags have been triggered. It would also be very nice to have a function in the PMX agent that lets us reset the flags, or a supervisor task that cycles power to the PMX using synaccess.

ykyohei, 2023-06-08:
Sound good. I started implementing them. We'll merge after testing it.
https://github.com/simonsobs/socs/tree/hwp_pmx

ykyohei, 2023-06-08:
We can query the error code and the error message, 
for example `(0, 'No error')`,  `(103,  'Operation denied while in PROTection state')`

Which level of monitoring is preferred in socs?
- publish only the error code to both influxDB and aggregator
- publish only the error code to both influxDB, error code and message to aggregator
- etc

closed 2023-06-21 by ykyohei