simonsobs/socs #462: HWP PMX agents stops collecting data when PMX is rebooted

jlashner, 2023-06-14:
When the voltage or current limits are hit on the PMX, we need to reboot the PMX using the synaccess agent. This will stop the `acq` process, and currently you need to restart the agent. It would be nice if the `acq` process was robust to short dropouts of the hardware.

ykyohei, 2023-06-22:
Do you think this issue is solved now? @jlashner 
Because we implemented a function to clear protection alarm without rebooting. https://github.com/simonsobs/socs/pull/463

jlashner, 2023-06-26:
Hi Kyohei, I'm happy to close this based on the changes in that PR! I think I may have also seen instances of the `acq` process not starting properly when the agent is first brought up, but if that continues to be the case I will start a new issue.