simonsobs/socs #470: tc400 agent throws error when `turn_turbo_off` task is run

msilvafe, 2023-06-30:
@ngalitzki noticed when he ran the `turn_turbo_off` task on the ocs-web interface an error was thrown and the session data for the task says `"Turbo unable to be turned off"` but the turbos appear to shut off just fine. Seems like a logging/agent error.

BrianJKoopman, 2023-06-30:
Interesting. It would be useful if there was a time we could debug this, which would involve turning the turbo on and off.

I'd like to see what the actual responses are to this line, if not at a lower level:
https://github.com/simonsobs/socs/blob/5899c242298a86d8539d5cfd155c1f071ffc8d0b/socs/agents/pfeiffer_tc400/drivers.py#L178

We do have [a test for this task](https://github.com/simonsobs/socs/blob/5899c242298a86d8539d5cfd155c1f071ffc8d0b/tests/integration/test_pfeiffer_tc400_agent_integration.py#L108-L119) with mock communication, so either we have our expected responses wrong, or the TC400 is sending something unexpected.

ngalitzki, 2023-06-30:
Running another LC set over the weekend, probably warm up sometime next week. Once the mix is fully recovered we can turn the turbos on and off to our hearts content.

mjrand, 2023-06-30:
I second the idea that the turbo is sending something unexpected back. I feel like we saw this issue once before at UCSD a long time ago where the agent would unexpectedly crash because the turbo responded something weird during acq. I could never consistently reproduce the error, but maybe we'll be able to reproduce this if we turn it on and off enough next week.

ngalitzki, 2023-07-03:
@BrianJKoopman  mix is recovered and turbos are in a good vacuum so now is a good time to debug this.

sanahabhimani, 2023-07-10:
I pinged Michael separately, but just wanted to note that whenever there's another good time to test, feel free to holler at me if Brian's too busy.

kmharrington, 2023-09-25:
Running into this again at the site. Would be nice to not have to worry about getting the turbos off.

BrianJKoopman, 2023-09-25:
I still need the details to that device response above. Lemme add some debug lines and we can run that at site so I can get the info I need from it next time this happens.

jlashner, 2023-09-25:
ya for now we can just add a more informative error message

mjrand, 2025-05-28:
This very long-standing and annoying bug seems to have been solved by @samdayweiss . Apparently the TC400 responds with a `000000` when the serial command to turn the turbo motor off succeeds in the `turn_turbo_motor_off` function. So when the command succeeded, the TC400 would send a `000000` and the function would return `response==111111`, which always returned `False`. So the agent would always see the function as failing and would raise the alarm for no reason.

PR  #887 Fixes this issue. Probably.