simonsobs/socs #481: pysnmp-lextudio doesn't support twisted

BrianJKoopman, 2023-07-20:
In #439 we replaced `pysnmp` with `pysnmp-lextudio`, however in debugging #478 I found that `pysnmp` is still getting installed in our Docker builds, as it's being pulled in as a dependency of `thola-snmpsim`.

`thola-snmpsim` is installed via the [testing requirements file](https://github.com/simonsobs/socs/blob/main/requirements/testing.txt).

This seems to be the only reason our snmp functionality still works, since `pysnmp-lextudio` doesn't have the module 'pysnmp.hlapi.twisted', which we [rely on](https://github.com/simonsobs/socs/blob/c126f2ecc872b3bd316f793322a69c9d01ee6812/socs/snmp.py#L4).

In a clean virtual environment if you run:
```
$ pip install pysnmp-lextudio
$ python
>>> import pysnmp.hlapi.twisted
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pysnmp.hlapi.twisted'
>>> pysnmp.__version__
'5.0.28'
```

BrianJKoopman, 2023-07-20:
@davidvng, were there other forks of pysnmp that might have twisted support? If not, perhaps best to move back to pysnmp for now. Though I'd be interested to hear if you think this partially reverted #439 still might solve #434, or if we'd just be causing ourselves more issues.

davidvng, 2023-07-21:
@BrianJKoopman can't find any well-used forks with twisted. I think reverting to `pysnmp` and keeping `pyasn1==0.4.8` should be fine.

closed 2023-07-31 by BrianJKoopman