simonsobs/socs #506: Need suprsync task to request agent to add all TcDirs

jlashner, 2023-08-16:
Right now the suprsync module has a function `create_all_timecode_dirs` which creates the TimecodeDir entries in the db for all files if it does not already exist. However, this is never called by the agent, so in the case the TimecodeDir for a file doesn't exist for whatever reason, we need to manually run this.

We should have a task in the suprsync agent that requests that the main process runs `create_all_timecode_dirs` on its next iteration.

kmharrington, 2023-09-05:
While working on that, we should also have a task that retries copying all files that were over their copy attempt limit. That will probably be the easiest way to fix finalization time errors or when something is just hanging. 

And another one that lets us request a list of all files with non-matching check sums and then another to set a file or list of files to ignore.

BrianJKoopman, 2026-01-21:
After https://github.com/simonsobs/socs/pull/564 was merged, `create_all_timecode_dirs` gets called every 6 hours:
https://github.com/simonsobs/socs/blob/91a4306d71e2e3b616c63d18eb6c361855ca53e4/socs/agents/suprsync/agent.py#L177-L179

I think this resolves this issue. Feel free to re-open/comment if that isn't the case.

closed 2026-01-21 by BrianJKoopman