simonsobs/socs #531: Vantage agent changes

jlashner, 2023-09-23:
A few changes and additions we'd like in the vantage agent based on current usage: 
- UVI is off by a factor of ten
- UVI overflows if it goes above 127. We should have the agent automatically convert negative UVIs to the value they would be if they were an unsigned int.
- Temp is in fahrenheit but we'd like celsius
- We'd like an estimate of the wind-chill / wind-chill-temp. We may be able to just do this in grafana with existing data

BrianJKoopman, 2023-10-11:
Mostly solved by #539. If we're fine with estimating wind-chill / wind-chill-temp in Grafana this can be closed.

BrianJKoopman, 2024-03-01:
Wind chill was added in #550. I had to revert the UVI factor of 10 fix in #636, due to the fact that since an integer was already published, InfluxDB rejects the new float values. I think it would be fine to continue to publish integers and just divide by 10 for display in Grafana. Let me know if that's still a problem.

closed 2026-03-04 by BrianJKoopman