simonsobs/socs #568: Fuel tank fill sensor Agent

BrianJKoopman, 2023-11-13:
There is some serial device that reads the fuel level of the main fuel tanks at site. This needs an Agent written for it. The start of some driver code has been written, get in touch for details if you're interested in working on this agent.

If possible, more information about the device should be posted here.

Daniel5615, 2023-11-13:
The monitoring is between Raspberry Pi B2 (pi-wheather-site) and USB to serial port (ttyUSB1). This are monitoring the large tanks that feed the generators; the comunications is bidirectional, where the script (path /home/user/Scripts/fueltanks_monitor.py), ask to serial port connected to tanks monitor device, that include a touchscreen, but this can´t trasfer directly the information of the level sensors to the computer server of monitoring data as Grafana or OCS; for this reason the reply of the tank monitor is in the ttyUSB1 in the pi-weather-site and this data should be included in monitoring data and is necesary create a OCS Agent for the large fuel tanks and obtaing this readings.

Hardware information:

No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 000f
Serial          : 00000000c3eb8dc1
Model           : Raspberry Pi Model B Rev 2
Eth0 MAC        : b8:27:eb:eb:8d:c1
Eth0 IPV4       : 192.168.0.128
Eth0 IPV6       : fe80::ecf0:80c:5781:7ff7

/dev/ttyUSB1
Bus 001 Device 005: ID 10c4:ea61 Silicon Labs CP210x UART Bridge

BrianJKoopman, 2023-11-13:
Thanks for posting this info. I was more wondering about the device at `/dev/ttyUSB1` -- the tank monitor device -- any info about that?

Daniel5615, 2023-11-13:
Only we have the information shared by Delwin Johnson, where he say us that the model installed haven't the propiety to share data between ethernet output; I attach now whit this comment the information available.
[GLT622 Protocol.pdf](https://github.com/simonsobs/socs/files/13340976/GLT622.Protocol.pdf)
[GLT622 user manual console.pdf](https://github.com/simonsobs/socs/files/13340977/GLT622.user.manual.console.pdf)