simonsobs/socs #631: `run_and_validate()` in HWP Supervisor doesn't check command ran successfully

BrianJKoopman, 2024-02-15:
I noticed that `run_and_validate()` doesn't actually "validate" that the command ran successfully, just that the initial OCS call goes through.

As written, it looks like Tasks could fail and the action would just move on to the next task call. `session['success']` would need to be checked to "validate" each task is complete.

_Originally posted by @BrianJKoopman in https://github.com/simonsobs/socs/pull/628#discussion_r1488300484_

ykyohei, 2024-04-16:
Once we fix this issue, I think I can test this in satp3 by intentionally turn off one of the hwp-agent and run the supervisor task.

jlashner, 2024-04-17:
Hi Kyohei, do you mind posting the code with how you are running supervisor tasks?

If you are running it through a hwp-supervisor client as opposed to through sorunlib, you need to manually check if the session finished successfully, since a failed agent operation will not otherwise raise an error in the client script. For instance, everything in sorunlib uses this [check_response](https://github.com/simonsobs/sorunlib/blob/7170fb9b62e8c29adb34912dbbd86f9521e672eb/src/sorunlib/_internal.py#L30) function to make sure the task finished successfully.

If you are already using sorunlib directly, than we need to figure out why this is not failing, since I'm pretty sure when an operation enters an Error state like in your examples above, `session['success']` should already be false.

ykyohei, 2024-04-17:
Hi Jack, I was running through hwp-supervisor client as follows.
```
sup = OCSClient('hwp-supervisor')

sup.pid_to_freq(target_freq = 2.0)
sup.pmx_off()
```

I actually didn't know that it is possible to run these from sorunlib. I will try again through sorunlib.
https://github.com/simonsobs/sorunlib/blob/7170fb9b62e8c29adb34912dbbd86f9521e672eb/src/sorunlib/hwp.py
If I run these commands through sorunlib, and if one of the agent operations fails, should it raise an error and stop the schedule?

I'm confused about the point of this issue now. We should run the supervisor task through sorunlib, and as long as we do so, this issue that Brian points out is not a real issue?

Also, do you think we should implement gripper.grip, gripper.ungrip, supervisor.enable_driver_board, and supervisor.disable_driver_board in sorunlib for the same reason?

jlashner, 2024-04-18:
yep, asserting session['success'] should work!

Brian is correctly pointing out that this needs to be asserted wherever client operations are called, but the hwp-supervisor is not checking when it calls out to other agents.

This needs to be done both in the supervisor agent, and also any client script running in nextline.

jlashner, 2024-04-18:
I think implementing these functions in sorunlib is a good idea to minimize the possibility of people making mistakes when calling them, but it isn't strictly necessary to get it to run.