simonsobs/socs #721: Make Agents robust to connection dropouts

BrianJKoopman, 2024-08-13:
Related to discussion https://github.com/simonsobs/socs/discussions/538, we need to make sure agents are robust against connection dropouts, whether that's a network interruption, serial connection dropout, or otherwise.

I'm making a single issue for this to avoid spamming 50+ individual issues. If you would like to contribute, read on! 

## Contributing
If you would like to work on "robustifying" an agent, please create an issue with a title like "Make \<AGENT NAME\> robust to connection dropouts" and assign yourself. This will help us keep track of which agents are actively being worked on, and which are up for grabs.

When you're ready, PR your code changes and [link the associated issue](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/using-keywords-in-issues-and-pull-requests). Once merged I'll update the checklist below.

(If you know any of the agents to already be robust, please comment here.)

## Robust Agents Checklist
* [ ] ACTiCameraAgent
* [ ] ACUAgent
* [ ] BlueforsAgent
* [ ] CrateAgent
* [ ] CryomechCPAAgent
* [ ] FPGAAgent
* [ ] FlowmeterAgent
* [ ] FTSAerotechAgent
* [ ] GeneratorAgent
* [ ] Hi6200Agent
* [ ] HWPBBBAgent
* [ ] HWPGripperAgent
* [ ] HWPPCUAgent
* [ ] HWPPicoscopeAgent
* [ ] HWPPIDAgent
* [ ] HWPPMXAgent
* [ ] HWPSupervisor
* [ ] ibootbarAgent
* [ ] LabJackAgent
* [ ] Lakeshore240Agent
* [ ] Lakeshore336Agent
* [ ] Lakeshore370Agent
* [ ] Lakeshore372Agent
* [ ] Lakeshore425Agent
* [ ] LATRtXYStageAgent
* [ ] MagpieAgent
* [ ] MeinbergM1000Agent
* [ ] MeinbergSyncboxAgent
* [ ] PfeifferAgent
* [ ] PfeifferTC400Agent
* [ ] PysmurfController
* [ ] PysmurfMonitor
* [ ] ScpiPsuAgent
* [ ] SmurfFileEmulator
* [ ] SmurfStreamSimulator
* [ ] SmurfTimingCardAgent
* [ ] SupRsync
* [ ] SynaccessAgent
* [ ] SynthAgent
* [ ] TektronixAWGAgent
* [ ] ThorlabsMC2000BAgent
* [ ] UCSCRadiometerAgent
* [ ] UPSAgent
* [ ] VantagePro2Agent
* [ ] WiregridActuatorAgent
* [ ] WiregridEncoderAgent
* [ ] WiregridKikusuiAgent
* [ ] WiregridTiltSensorAgent

BrianJKoopman, 2024-08-14:
It sounds like there might be some hardware related issues contributing, but the `HWPPIDAgent` is one of the commonly crashing agents, and a good one to take a look at next for anyone interested. See https://github.com/simonsobs/chwp-discussions/discussions/21.

davidvng, 2024-09-17:
@BrianJKoopman the `ibootbarAgent` and the `UPSAgent` should already robust to connection dropouts