simonsobs/socs #796: HWP PCU agent crash

BrianJKoopman, 2024-11-21:
The HWP PCU agent has crashed on satp3 several times with the below error.

**socs docker tag:** v0.5.0

```
2024-11-21 17:26:30.373 Args: ['--instance-id', 'hwp-pcu']
2024-11-21 17:26:30.615 Installed OCS Plugins: ['socs', 'ocs']
2024-11-21 17:26:31.024 Renaming this process to: "ocs-agent:hwp-pcu"
2024-11-21 17:26:31.136 2024-11-21T17:26:31+0000 Using OCS version 0.11.0
2024-11-21 17:26:31.136 2024-11-21T17:26:31+0000 ocs: starting <class 'ocs.ocs_agent.OCSAgent'> @ satp3.hwp-pcu
2024-11-21 17:26:31.136 2024-11-21T17:26:31+0000 log_file is apparently None
2024-11-21 17:26:31.163 2024-11-21T17:26:31+0000 transport connected
2024-11-21 17:26:31.168 2024-11-21T17:26:31+0000 session joined: {'authextra': {'x_cb_node': '4e2f32e9b379-7',
2024-11-21 17:26:31.168                'x_cb_peer': 'tcp4:192.168.13.42:39672',
2024-11-21 17:26:31.168                'x_cb_pid': 14,
2024-11-21 17:26:31.168                'x_cb_worker': 'worker001'},
2024-11-21 17:26:31.168  'authid': 'R9HR-S6LM-JEGQ-CAKK-R3AA-ARV3',
2024-11-21 17:26:31.168  'authmethod': 'anonymous',
2024-11-21 17:26:31.168  'authprovider': 'static',
2024-11-21 17:26:31.168  'authrole': 'iocs_agent',
2024-11-21 17:26:31.168  'realm': 'test_realm',
2024-11-21 17:26:31.168  'resumable': False,
2024-11-21 17:26:31.168  'resume_token': None,
2024-11-21 17:26:31.168  'resumed': False,
2024-11-21 17:26:31.168  'serializer': 'cbor.batched',
2024-11-21 17:26:31.168  'session': 5902466971262715,
2024-11-21 17:26:31.168  'transport': {'channel_framing': 'websocket',
2024-11-21 17:26:31.168                'channel_id': {},
2024-11-21 17:26:31.168                'channel_serializer': None,
2024-11-21 17:26:31.168                'channel_type': 'tcp',
2024-11-21 17:26:31.168                'http_cbtid': None,
2024-11-21 17:26:31.168                'http_headers_received': None,
2024-11-21 17:26:31.168                'http_headers_sent': None,
2024-11-21 17:26:31.168                'is_secure': False,
2024-11-21 17:26:31.168                'is_server': False,
2024-11-21 17:26:31.168                'own': None,
2024-11-21 17:26:31.168                'own_fd': -1,
2024-11-21 17:26:31.168                'own_pid': 6,
2024-11-21 17:26:31.168                'own_tid': 6,
2024-11-21 17:26:31.168                'peer': 'tcp4:192.168.0.223:8005',
2024-11-21 17:26:31.168                'peer_cert': None,
2024-11-21 17:26:31.168                'websocket_extensions_in_use': None,
2024-11-21 17:26:31.168                'websocket_protocol': None}}
2024-11-21 17:26:31.174 2024-11-21T17:26:31+0000 startup-op: launching main
2024-11-21 17:26:31.174 2024-11-21T17:26:31+0000 start called for main
2024-11-21 17:26:31.174 2024-11-21T17:26:31+0000 main:0 Status is now "starting".
2024-11-21 17:26:31.175 2024-11-21T17:26:31+0000 main:0 Status is now "running".
2024-11-21 17:26:31.176 2024-11-21T17:26:31+0000 Connected to PCU
2024-11-21 17:26:32.177 2024-11-21T17:26:32+0000 Cleared buffer
2024-11-21 17:26:35.784 2024-11-21T17:26:35+0000 Status is failed, cleared buffer
2024-11-21 17:30:36.660 2024-11-21T17:30:36+0000 main:0 CRASH: [Failure instance: Traceback: <class 'serial.serialutil.SerialException'>: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2024-11-21 17:30:36.660 /usr/lib/python3.8/threading.py:932:_bootstrap_inner
2024-11-21 17:30:36.660 /usr/lib/python3.8/threading.py:870:run
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/twisted/_threads/_threadworker.py:49:work
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/twisted/_threads/_team.py:192:doWork
2024-11-21 17:30:36.660 --- <exception caught here> ---
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:269:inContext
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/twisted/python/threadpool.py:285:<lambda>
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/twisted/python/context.py:117:callWithContext
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/twisted/python/context.py:82:callWithContext
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/ocs/ocs_agent.py:984:_running_wrapper
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/socs/agents/hwp_pcu/agent.py:128:main
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/socs/agents/hwp_pcu/agent.py:88:_get_and_publish_data
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/socs/agents/hwp_pcu/drivers/hwp_pcu.py:93:get_status
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/socs/agents/hwp_pcu/drivers/hwp_pcu.py:62:relay_read
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/socs/agents/hwp_pcu/drivers/hwp_pcu.py:39:read
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/serial/serialutil.py:663:read_until
2024-11-21 17:30:36.660 /usr/local/lib/python3.8/dist-packages/serial/serialposix.py:595:read
2024-11-21 17:30:36.660 ]
2024-11-21 17:30:36.660 2024-11-21T17:30:36+0000 main:0 Status is now "done".
2024-11-21 17:35:43.953 2024-11-21T17:35:43+0000 caught SIGTERM!
2024-11-21 17:35:43.953 2024-11-21T17:35:43+0000 Stopping all running sessions
2024-11-21 17:35:43.953 2024-11-21T17:35:43+0000 Stopping session main
2024-11-21 17:35:46.954 2024-11-21T17:35:46+0000 stopping reactor
2024-11-21 17:35:46.958 2024-11-21T17:35:46+0000 session left: CloseDetails(reason=<wamp.close.normal>, message='None')
2024-11-21 17:35:46.958 2024-11-21T17:35:46+0000 Stopping all running sessions
2024-11-21 17:35:46.959 2024-11-21T17:35:46+0000 Stopping session main
2024-11-21 17:35:49.965 2024-11-21T17:35:49+0000 transport disconnected
2024-11-21 17:35:49.965 2024-11-21T17:35:49+0000 waiting for reconnection
2024-11-21 17:35:49.965 2024-11-21T17:35:49+0000 waiting at least 9.99999713897705 more seconds before giving up
2024-11-21 17:35:49.965 2024-11-21T17:35:49+0000 Scheduling retry 1 to connect <twisted.internet.endpoints.TCP4ClientEndpoint object at 0x7f05102ab400> in 2.1901233981771937 seconds.
2024-11-21 17:35:49.966 2024-11-21T17:35:49+0000 Main loop terminated.
```

ykyohei, 2024-11-21:
Is the dynamic reconnection discussed in https://github.com/simonsobs/socs/discussions/538 still recommended as an approach to improve robustness for this type of error?

BrianJKoopman, 2024-11-22:
> Is the dynamic reconnection discussed in #538 still recommended as an approach to improve robustness for this type of error?

That discussion predates the "actions" structure that @jlashner implemented here (in #600.) While the implementation in that discussion would still work, and make the drivers robust (if ever used separately from the agent), it is maybe simpler to add some error handling to these few lines:
https://github.com/simonsobs/socs/blob/10b48d193e845b8ef7a1c5c6ae498c6bb56b7e3a/socs/agents/hwp_pcu/agent.py#L127-L131

Wrapping that in a try/except to handle the raised error and set `PCU = None`, then these few lines handle the reconnection for you:
https://github.com/simonsobs/socs/blob/10b48d193e845b8ef7a1c5c6ae498c6bb56b7e3a/socs/agents/hwp_pcu/agent.py#L114-L125

@jlashner should comment if that's not what he had in mind with this structure.

ykyohei, 2024-11-22:
Thanks, I started to try improving this here. https://github.com/simonsobs/socs/pull/797
I'm testing this at satp3 now.

BrianJKoopman, 2024-11-25:
Resolved in #797.

closed 2024-11-25 by BrianJKoopman