simonsobs/socs #842: hwp-synaccess turns on all the channels after power cycle

ykyohei, 2025-03-21:
I noticed that hwp-synaccess turns on all the channels after it is power cycled.
This is not an ideal behavior and we would like to be able to disable certain outlet channels.

ykyohei, 2025-03-24:
Synaccess Technical support says, following. We can configure this from web UI, the default ip address is 192.168.1.100.

> In the Outlet Setup section of the web UI, you will find a column named "Power Up Initial State". This is the initial state of the outlet upon the PDU powering up. You can either set this option to OFF (so that the outlet is always off when the PDU first starts up) here.

BrianJKoopman, 2025-03-24:
> > In the Outlet Setup section of the web UI, you will find a column named "Power Up Initial State". This is the initial state of the outlet upon the PDU powering up. You can either set this option to OFF (so that the outlet is always off when the PDU first starts up) here.

Is there a "last state" option? Or is the power up initial state either "on" or "off"?

ykyohei, 2025-04-05:
I've changed initial states of outlets of hwp driverboards on all SATs, so I will close this.

closed 2025-04-05 by ykyohei