simonsobs/socs #866: Suprsync needs to smartly delete certain smurf files

kmharrington, 2025-04-29:
We have the timestream suprsyncs set to delete files after a certain amount of time but not the smurf ones. This is because we don't want to delete tune files or the last bg maps, bias steps, etc. This means that the smurf servers slowly get full off of these sorts of files and, for the LAT, this is happening faster than we'd like. 

We should upgrade the suprsync delete logic so that we can delete older smurf files that we know we won't need.

BrianJKoopman, 2025-04-30:
With Jack gone I think we're going to need more info in this issue before someone can implement it. The more detailed the better. What files should be deleted? What needs to stay? (It sounds like we never want to delete tune files?) Example paths, lists of files, etc would all be helpful.

BrianJKoopman, 2025-05-01:
> One trouble is I don't think we have a place to test before deploying any changes? I'm not sure how much of this was implemented in the E2E system.

Right, the E2E system needs some attention. I hoping to get that working again soon, as there are several other items that it'd be helpful for right now. I took a quick look and there's not any simulated data left on it that I could find, but files relevant for this are output by the [smurf file emulator](https://github.com/simonsobs/socs/blob/main/socs/agents/smurf_file_emulator/agent.py).

kmharrington, 2025-05-01:
Actually it looks like the file emulator does enough for this on the file side. So we should be able to test a setup there if that gets stood back up.