simonsobs/socs #867: ACU: Crash at end of boresight command

BrianJKoopman, 2025-04-30:
On satp1 this morning I aborted a schedule in preparation for site hand off. I sent the telescope to 180, 60, spun down the HWP, then rotated to -30 in boresight all via ocs-web. The rotation seemed to work, it made it to -30, but at the end errored out with:
```
2025-04-30T13:18:07+0000 start called for go_to
2025-04-30T13:18:07+0000 go_to:701 Status is now "starting".
2025-04-30T13:18:07+0000 go_to:701 Status is now "running".
2025-04-30T13:18:07+0000 Clearing faults to prepare for motion.
2025-04-30T13:18:08+0000 Requested position: az=180.0, el=60.0
2025-04-30T13:18:08+0000 Azimuth.state=INIT             dt=  0.000 dist=  28.500
2025-04-30T13:18:09+0000 Azimuth.state=WAIT_MOVING      dt=  0.101 dist=  28.490
2025-04-30T13:18:09+0000 Azimuth.state=WAIT_STILL       dt=  0.202 dist=  28.476
2025-04-30T13:18:09+0000 Ignoring requested motion on Elevation
2025-04-30T13:18:29+0000 Azimuth.state=DONE             dt= 20.518 dist=   0.004
2025-04-30T13:18:30+0000 go_to:701 Move complete.
2025-04-30T13:18:30+0000 go_to:701 Status is now "done".
2025-04-30T13:18:30+0000 Azimuth_mode is now "Stop"
2025-04-30T13:31:40+0000 start called for set_boresight
2025-04-30T13:31:40+0000 set_boresight:702 Status is now "starting".
2025-04-30T13:31:40+0000 set_boresight:702 Status is now "running".
2025-04-30T13:31:40+0000 Clearing faults to prepare for motion.
2025-04-30T13:31:41+0000 Commanded position: boresight=-30.0
2025-04-30T13:31:41+0000 Boresight.state=INIT           dt=  0.000 dist=  30.000
2025-04-30T13:31:41+0000 Boresight.state=WAIT_MOVING    dt=  0.101 dist=  30.000
2025-04-30T13:31:41+0000 Boresight.state=WAIT_STILL     dt=  0.224 dist=  30.000
2025-04-30T13:31:52+0000 set_boresight:702 CRASH: [Failure instance: Traceback: <class 'twisted.internet.defer.CancelledError'>:
/opt/venv/lib/python3.10/site-packages/twisted/internet/defer.py:2013:_inlineCallbacks
/opt/venv/lib/python3.10/site-packages/twisted/python/failure.py:467:throwExceptionIntoGenerator
/opt/venv/lib/python3.10/site-packages/socs/agents/acu/agent.py:1483:set_boresight
/opt/venv/lib/python3.10/site-packages/twisted/internet/defer.py:2013:_inlineCallbacks
/opt/venv/lib/python3.10/site-packages/twisted/python/failure.py:467:throwExceptionIntoGenerator
/opt/venv/lib/python3.10/site-packages/socs/agents/acu/agent.py:1299:_go_to_axis
]
2025-04-30T13:31:52+0000 set_boresight:702 Status is now "done".
```

I did not check "Set mode=Stop at end?". This was with image tag: `v0.5.2-49-gd23277f-dev`

mhasself, 2025-05-01:
This is very odd and does not appear to be a simple bug in the code.  The move was initiated and was in progress and being monitored as normal (several similar boresight moves succeeded, before and since this one).  The "CancelledError" occurs on a line in go_to_axis that is just `yield dsleep(0.1)`.  How anyone would get to cancel that is very unclear.  There's nothing else interesting in the logs.

BrianJKoopman, 2025-05-01:
Thanks for investigating. I'll leave this open for now and see if it comes up again. If it doesn't, I expect we'll close this as unable to reproduce.