simonsobs/socs #882: hwp-supervisor: increasing timeout on `grip` command

ktcrowley, 2025-05-20:
Command to grip through `hwp-supervisor` seems to often fail with an error, but the `hwp-gripper` agent has not thrown and error and continues to complete grip. Perhaps this is OK from a safety perspective given the supervisor only passes the command to the gripper agent if it's safe to do so, but it is a confusing issue for remote users and, because it's not clear the process is still running, there's a minor risk that the next command to `hwp-supervisor` is not compatible with the assumed gripper state.

@bbixler500 mentioned that this is likely just due to inter-agent communication blips, so hopefully this is a quick fix whenever it can be edited + tested.

ktcrowley, 2025-05-20:
Here ya go:
`2025-05-20T18:34:30+0000 Setting state: ControlState.Error(traceback='Traceback (most recent call last):\n  File "/opt/venv/lib/python3.10/site-packages/socs/agents/hwp_supervisor/agent.py", line 1299, in update\n    self.run_and_validate(clients.gripper.grip)\n  File "/opt/venv/lib/python3.10/site-packages/socs/agents/hwp_supervisor/agent.py", line 1121, in run_and_validate\n    raise ControlClientError("op-wait timed out")\nocs.client_http.ControlClientError: op-wait timed out\n', start_time=1747766070.9647164)`

BrianJKoopman, 2025-05-20:
Thanks, the default timeout is 30 seconds. The grip operation that caused this timeout took nearly 5 minutes:
```
2025-05-20T18:34:00+0000 start called for grip
2025-05-20T18:34:00+0000 grip:5 Status is now "starting".
2025-05-20T18:34:00+0000 grip:5 Status is now "running".
2025-05-20T18:38:39+0000 grip:5 Finished gripping procedure
2025-05-20T18:38:39+0000 grip:5 Status is now "done".
```

What is the expected duration of a grip? @bbixler500

bbixler500, 2025-05-20:
It depends on the specific SAT (i.e. what that physical grip distance is and how the starting position is defined) but should be somewhere between 5-10 minutes

BrianJKoopman, 2025-05-20:
Great, does ungrip have the same problem? How long do those usually take? I see it took 28 seconds most recently.

I'm going to put a 12 minute timeout on the grip command. Seems like we might want a slightly longer than 30 second timeout for ungrip, is 60 seconds reasonable?

bbixler500, 2025-05-20:
Ungrip is much faster as it doesn't need to slowly inch inwards to reach the limit switches. 60 seconds sounds like a reasonable timeout

PR #883 (2025-05-20): hwp-supervisor: Set longer timeout for grip commands

## Description

This sets a longer timeout for the grip and ungrip commands to execute (12 minutes, and 1 minute, respectively) from the supervisor agent.

## Motivation and Context

Resolves #882.

## How Has This Been Tested?

Not yet tested.

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## Checklist:

- [x] My code follows the code style o [... cut]
closed 2025-05-20 by BrianJKoopman