simonsobs/socs #904: TC400 agent error code datatype is incorrect

mjrand, 2025-07-22:
The TC400 agent has a data stream in acq for the error codes read from the pfeiffer TC400 turbo controllers. Currently, the driver reads the error code as a string and also passes it into the data stream as a string. This has made parsing the error code on grafana slightly difficult.

I think we need to update the code so that it also casts the error code as an int so we can display it easily in a new data field.

This can be tested in between satp2 receiving power again and satp2 pumping out with the turbos.

BrianJKoopman, 2025-07-22:
Thanks @mjrand, are the error codes always ints? The example in the `acq` process' `session.data` lists `Err001` as an example. Would you just drop the 'Err'?

mjrand, 2025-07-25:
I've just looked at the TC400 manual. It looks like in the case of "no error" the turbo will return "000000" which could be blindly cast to an int.

But when there is an error it will actually return a string "Err001" like you gave. It can also return something like "Wrn007" as an error. I'm not sure what the difference is... but it seems like "Err045" and "Wrn045" both mean "high temperature motor" for instance. Maybe we can just strip the "Err"/"Wrn" from the error code and publish the 45 to grafana? As long as the error numbers aren't degenerate that could work?

BrianJKoopman, 2025-08-01:
The distinction seem likely to be different thresholds for "warning" vs "error", though the manual doesn't seem to specify what those thresholds are.

Looking at the error codes table, while 045 are similar between Wrn/Err, that isn't always true, for instance `Err007` is "Operating fluid deficiency" and `Wrn007` is "Low voltage / mains power failure". So I'm not convinced it's the best idea to strip the string and cast these to a float.

> This has made parsing the error code on grafana slightly difficult.
I take it you're trying to view the history of these error codes? If you only care for the current state of the error code I'm pretty sure we can display that string.